sparkfun/SparkFun_STUSB4500_Arduino_Library #2: Does this support PPS?

jxltom, 2021-02-12:
Such as 5v-20v with 20mV increment

awende, 2021-02-12:
I haven't tested it, so I can't say for certain, but it might be able to. To my best understanding, PPS is a recent addition to the PD3.0 standard. The controller IC (STUSB4500) is certified for PD2.0, but interoperable with PD3.0.

The way the controller works is negotiation happens at power up or after a reset condition. If you had a power adapter that supported PPS, you would need to set the new voltage to the power delivery board using this library and then reset the board, so that the new contract can be negotiated with the PPS supported power adapter.

awende, 2021-02-19:
The STUSB4500 sends fixed PDO messages, and PPS is a separate message.

You can refer to this [link](https://community.st.com/s/question/0D50X0000BQPoId/how-do-i-configure-the-stusb4500-as-a-pps-sink) which lists a few of the ST product offerings that support PPS

closed 2021-02-19 by awende

jxltom, 2021-02-20:
@awende，thanks for the clarification!