sparkfun/SparkFun_STUSB4500_Arduino_Library #3: usb.write

infamy, 2021-02-23:
So this function is not of much use, there is no way to pass a new set of defaults to it to program the board without rewriting the function in the library.

awende, 2021-02-23:
Some context might show it's usefulness for the usb.write(DEFAULT):

The `DEFAULT` parameter was created out of necessity. During the development of the board, I ran into an issue where one or more of the NVM bytes would have the wrong value written to them. As a result, you could change the PDO settings, but the board would stay at 5V. If I copied the NVM map from a fresh IC, then it would start to work again, because not all of the bytes are modified by the set functions.

What would be the use case for passing the write function an array? With the set functions modifying the bits, and then calling usb.write(), that is writing a new set of values.

infamy, 2021-02-23:
Interesting, I'm running into this issue with my board. I can only get 5v out.

So the reason to write an array was to use the NVM map from the ST app. https://github.com/usb-c/STUSB4500 <- they have a GUI for making an NVM map.

infamy, 2021-02-23:
I should add, that i have noticed if i write anything (change a setting then write) and then try and write defaults, it does not seem to take.

awende, 2021-02-23:
Can you post the code that you're using?

infamy, 2021-03-08:
I update to the latest 1.1.3. After run it, i loose all connection to the STUSB, i have more board but would rather not brick any more.
See attached for what was run. Ignore the block  of code commented out, have not used that in a while.
[USBC_PPv1.txt](https://github.com/sparkfun/SparkFun_STUSB4500_Arduino_Library/files/6099691/USBC_PPv1.txt)

awende, 2021-03-08:
I uploaded the code you sent using my Uno, and didn't have an issue. So I can try to replicate your setup, what kind of microcontroller are you using?

The begin function, is only checking for an acknowledgment from a device at the specified address, which should be 0x28. You could try running the [I2C Scanner](https://playground.arduino.cc/Main/I2cScanner/) sketch to see what addresses the board is able to access.

infamy, 2021-03-08:
I'm on an itsy bitsy m0.

Will run the i2c scanner. it communicated fine till i tried the write default, which before was doing nothing...

awende, 2021-03-12:
I wasn't able to replicate that problem, but in testing I did notice that V1.1.3 wasn't reading PDO values correctly. and pushed the changes to V1.1.4. I tested all of the examples on a wide variety of boards, including the SAMD21, which the itsy bitsy m0 uses and did not run into any issues.