sparkfun/SparkFun_STUSB4500_Arduino_Library #4: usb.setVoltage input does not match usb.getVoltage output

jake, 2021-03-03:
I'm having an issue running the built in scripts to change the PDO voltage values. The values I pass to the `usb.setVoltage` function does not get returned by the `usb.getVoltage` function. Not sure if the issue is with set writing incorrectly or get reading incorrectly.

Sample code (which is just the bundled example):

```
/*
  Writing the Settings to the STUSB4500 Power Delivery Board
  By: Alex Wende
  SparkFun Electronics
  Date: February 6th, 2020
  License: This code is public domain but you buy me a beer if you use this and we meet someday (Beerware license).
  Feel like supporting our work? Buy a board from SparkFun!
  https://www.sparkfun.com/products/15801

  This example demonstrates how to change the NVM settings saved to the STUSB4500

  Quick-start:
  - Use a SparkFun RedBoard Qwiic -or- attach the Qwiic Shield to your Arduino/Photon/ESP32 or other
  - Uncomment and modify the variables of the functions you want to change
  - Upload the sketch
  - Plug the Power Delivery Board onto the RedBoard/shield
  - Open the serial monitor and set the baud rate to 115200
  - The RedBoard will connect to the Power Delivery Board over I2C and print out all of the settings saved.
*/
// Include the SparkFun STUSB4500 library.
// Click here to get the library: http://librarymanager/All#SparkFun_STUSB4500

#include <Wire.h>
#include <SparkFun_STUSB4500.h>

STUSB4500 usb;

void setup()
{
  Serial.begin(115200);
  Wire.begin(); //Join I2C bus

  delay(500);

  /* The Power Delivery board uses the default settings with address 0x28 using Wire.

     Opionally, if the address jumpers are modified, or using a different I2C bus,
     these parameters can be changed here. E.g. usb.begin(0x29,Wire1)

     It will return true on success or false on failure to communicate. */
  if (!usb.begin())
  {
    Serial.println("Cannot connect to STUSB4500.");
    Serial.println("Is the board connected? Is the device ID correct?");
    while (1);
  }

  Serial.println("Connected to STUSB4500!");
  delay(100);

  /* Set Number of Power Data Objects (PDO) 1-3 */
  usb.setPdoNumber(2);

  /* PDO1
    - Voltage fixed at 5V
    - Current value for PDO1 0-5A, if 0 used, FLEX_I value is used
    - Under Voltage Lock Out (setUnderVoltageLimit) fixed at 3.3V
    - Over Voltage Lock Out (setUpperVoltageLimit) 5-20%
  */
  //  usb.setCurrent(1,0.5);
  //  usb.setUpperVoltageLimit(1,20);

  /* PDO2
    - Voltage 5-20V
    - Current value for PDO2 0-5A, if 0 used, FLEX_I value is used
    - Under Voltage Lock Out (setUnderVoltageLimit) 5-20%
    - Over Voltage Lock Out (setUpperVoltageLimit) 5-20%
  */
  usb.setVoltage(2, 12.0);
  usb.setCurrent(2, 0.5);
  //  usb.setLowerVoltageLimit(2,20);
  //  usb.setUpperVoltageLimit(2,20);

  /* PDO3
    - Voltage 5-20V
    - Current value for PDO3 0-5A, if 0 used, FLEX_I value is used
    - Under Voltage Lock Out (setUnderVoltageLimit) 5-20%
    - Over Voltage Lock Out (setUpperVoltageLimit) 5-20%
  */
  //  usb.setVoltage(3,20.0);
  //  usb.setCurrent(3,0.5);
  //  usb.setLowerVoltageLimit(3,20);
  //  usb.setUpperVoltageLimit(3,20);

  /* Flexible current value common to all PDOs */
  //  usb.setFlexCurrent(1.0);

  /* Unconstrained Power bit setting in capabilities message sent by the sink */
  usb.setExternalPower(false);

  /* USB 2.0 or 3.x data communication capability by sink system */
  usb.setUsbCommCapable(false);

  /* Selects POWER_OK pins configuration
     0 - Configuration 1
     1 - No applicable
     2 - Configuration 2 (default)
     3 - Configuration 3
  */
  //  usb.setConfigOkGpio(2);

  /* Selects GPIO pin configuration
     0 - SW_CTRL_GPIO
     1 - ERROR_RECOVERY
     2 - DEBUG
     3 - SINK_POWER
  */
  //  usb.setGpioCtrl(3);

  /* Selects VBUS_EN_SNK pin configuration */
  //  usb.setPowerAbove5vOnly(false);

  /* In case of match, selects which operating current from the sink or the
     source is to be requested in the RDO message */
  usb.setReqSrcCurrent(true);

  /* Write and save settings to STUSB4500 */
  usb.write();

  /* Read the NVM settings to verify the new settings are correct */
  usb.read();

  Serial.println("New Parameters:\n");

  /* Read the Power Data Objects (PDO) highest priority */
  Serial.print("PDO Number: ");
  Serial.println(usb.getPdoNumber());

  /* Read settings for PDO1 */
  Serial.println();
  Serial.print("Voltage1 (V): ");
  Serial.println(usb.getVoltage(1));
  Serial.print("Current1 (A): ");
  Serial.println(usb.getCurrent(1));
  Serial.print("Lower Voltage Tolerance1 (%): ");
  Serial.println(usb.getLowerVoltageLimit(1));
  Serial.print("Upper Voltage Tolerance1 (%): ");
  Serial.println(usb.getUpperVoltageLimit(1));
  Serial.println();

  /* Read settings for PDO2 */
  Serial.print("Voltage2 (V): ");
  Serial.println(usb.getVoltage(2));
  Serial.print("Current2 (A): ");
  Serial.println(usb.getCurrent(2));
  Serial.print("Lower Voltage Tolerance2 (%): ");
  Serial.println(usb.getLowerVoltageLimit(2));
  Serial.print("Upper Voltage Tolerance2 (%): ");
  Serial.println(usb.getUpperVoltageLimit(2));
  Serial.println();

  /* Read settings for PDO3 */
  Serial.print("Voltage3 (V): ");
  Serial.println(usb.getVoltage(3));
  Serial.print("Current3 (A): ");
  Serial.println(usb.getCurrent(3));
  Serial.print("Lower Voltage Tolerance3 (%): ");
  Serial.println(usb.getLowerVoltageLimit(3));
  Serial.print("Upper Voltage Tolerance3 (%): ");
  Serial.println(usb.getUpperVoltageLimit(3));
  Serial.println();

  /* Read the flex current value */
  Serial.print("Flex Current: ");
  Serial.println(usb.getFlexCurrent());

  /* Read the External Power capable bit */
  Serial.print("External Power: ");
  Serial.println(usb.getExternalPower());

  /* Read the USB Communication capable bit */
  Serial.print("USB Communication Capable: ");
  Serial.println(usb.getUsbCommCapable());

  /* Read the POWER_OK pins configuration */
  Serial.print("Configuration OK GPIO: ");
  Serial.println(usb.getConfigOkGpio());

  /* Read the GPIO pin configuration */
  Serial.print("GPIO Control: ");
  Serial.println(usb.getGpioCtrl());

  /* Read the bit that enables VBUS_EN_SNK pin only when power is greater than 5V */
  Serial.print("Enable Power Only Above 5V: ");
  Serial.println(usb.getPowerAbove5vOnly());

  /* Read bit that controls if the Source or Sink device's
     operating current is used in the RDO message */
  Serial.print("Request Source Current: ");
  Serial.println(usb.getReqSrcCurrent());
}

void loop()
{
}
```

Output:

```
Connected to STUSB4500!
New Parameters:

PDO Number: 2

Voltage1 (V): 49.80
Current1 (A): 1.50
Lower Voltage Tolerance1 (%): 0
Upper Voltage Tolerance1 (%): 15

Voltage2 (V): 50.40
Current2 (A): 0.50
Lower Voltage Tolerance2 (%): 20
Upper Voltage Tolerance2 (%): 10

Voltage3 (V): 50.20
Current3 (A): 0.50
Lower Voltage Tolerance3 (%): 20
Upper Voltage Tolerance3 (%): 20

Flex Current: 2.00
External Power: 0
USB Communication Capable: 0
Configuration OK GPIO: 2
GPIO Control: 1
Enable Power Only Above 5V: 0
Request Source Current: 1
```

Possibly of note, I am using a Metro Mini from Adafruit to program the board. 

Arduino IDE 1.8.13
Mac 10.14.6 (18G95)
Library 1.1.0

awende, 2021-03-04:
Thank you for posting your set up along with the code and output, that greatly helped with identifying the issue quickly.

I published [release 1.1.1](https://github.com/sparkfun/SparkFun_STUSB4500_Arduino_Library/releases/tag/V1.1.1), which fixes this issue. If you're updating from Arduino, the library manager should be updated in a couple hours with this latest release.

The issue in V1.1.0 was a line of code which combined four 8-bit values into a single 32-bit value, which worked fine on a 32-bit MCU. But for an 8-bit controller, like what's used on the Metro Mini, the 2 MSB were returning 0xFF instead of the proper values. I updated the function that created the problem to first pass the 8-bit values into a 32-bit variable and then shift the bits into position, instead of creating a 32-bit value in a single line.

closed 2021-03-04 by awende

awende, 2021-03-04:
Oops, just realized that when I created the release, I forgot to update the version in the library.properties file. I updated my response to link the the correct release that Arduino's library manager will show and updated the release in github for consistency.

jake, 2021-03-05:
Awesome, glad I thought to share it was a Metro Mini.

Running the code I shared initially, it seems to work. Here is the new output:

```Connected to STUSB4500!
New Parameters:

PDO Number: 2

Voltage1 (V): 5.00
Current1 (A): 1.50
Lower Voltage Tolerance1 (%): 0
Upper Voltage Tolerance1 (%): 15

Voltage2 (V): 12.00
Current2 (A): 1.75
Lower Voltage Tolerance2 (%): 20
Upper Voltage Tolerance2 (%): 10

Voltage3 (V): 15.00
Current3 (A): 0.50
Lower Voltage Tolerance3 (%): 20
Upper Voltage Tolerance3 (%): 10

Flex Current: 2.00
External Power: 0
USB Communication Capable: 0
Configuration OK GPIO: 2
GPIO Control: 1
Enable Power Only Above 5V: 0
Request Source Current: 1
```

Interestingly, running the built-in SetDefaults (aka `usb.write(DEFAULT);`) results in this. Seems like possibly PDD2 is still using the incorrect values? Is this related to #3? Note that I ran this *before* the SetParamaters output above, after having already ran it a few times on the old library version.

```Connected to STUSB4500
Done writing default settings!

PDO Number: 2

Voltage1 (V): 5.00
Current1 (A): 1.50
Lower Voltage Tolerance1 (%): 0
Upper Voltage Tolerance1 (%): 15

Voltage2 (V): 50.40
Current2 (A): 0.50
Lower Voltage Tolerance2 (%): 20
Upper Voltage Tolerance2 (%): 10

Voltage3 (V): 15.00
Current3 (A): 0.50
Lower Voltage Tolerance3 (%): 20
Upper Voltage Tolerance3 (%): 10

Flex Current: 2.00
External Power: 0
USB Communication Capable: 0
Configuration OK GPIO: 2
GPIO Control: 1
Enable Power Only Above 5V: 0
Request Source Current: 0
```

awende, 2021-03-05:
I wasn't able to replicate the voltage this time for PDO2, but I was able to verify that the voltage, current and PDO priority number were not appearing to reset correctly.

Try out [version 1.1.3](https://github.com/sparkfun/SparkFun_STUSB4500_Arduino_Library/releases/tag/V1.1.3), that should fix the issue.

faircm2, 2023-11-28:
I have the following problem. I set the PDO values for PDO2 and PDO3 to 15V/3A and 20V/5A resp., using a modified version of Example2-SetParameters.ino:

`#include <Wire.h>
#include <SparkFun_STUSB4500.h>

STUSB4500 usb;
double voltage;
int pdo_number;

void setup() 
{
  Serial.begin(115200);
  Wire.begin(); //Join I2C bus
  
  delay(500);
  
  /* The Power Delivery board uses the default settings with address 0x28 using Wire.
  
     Opionally, if the address jumpers are modified, or using a different I2C bus,
     these parameters can be changed here. E.g. usb.begin(0x29,Wire1)
  
     It will return true on success or false on failure to communicate. */
  if(!usb.begin())
  {
    Serial.println("Cannot connect to STUSB4500.");
    Serial.println("Is the board connected? Is the device ID correct?");
    while(1);
  }
  
  Serial.println("Connected to STUSB4500!");
  delay(100);

  /* Set Number of Power Data Objects (PDO) 1-3 */
  usb.setPdoNumber(3);

  /* PDO1
   - Voltage fixed at 5V
   - Current value for PDO1 0-5A, if 0 used, FLEX_I value is used
   - Under Voltage Lock Out (setUnderVoltageLimit) fixed at 3.3V
   - Over Voltage Lock Out (setUpperVoltageLimit) 5-20%
  */
  usb.setCurrent(1, 1.5);
  usb.setUpperVoltageLimit(1, 5);

  /* PDO2
   - Voltage 5-20V
   - Current value for PDO2 0-5A, if 0 used, FLEX_I value is used
   - Under Voltage Lock Out (setUnderVoltageLimit) 5-20%
   - Over Voltage Lock Out (setUpperVoltageLimit) 5-20%
  */
  usb.setVoltage(2, 15.0); //10
  usb.setCurrent(2, 3.0); //5.0
  usb.setLowerVoltageLimit(2, 20);
  usb.setUpperVoltageLimit(2, 20);

  /* PDO3
   - Voltage 5-20V
   - Current value for PDO3 0-5A, if 0 used, FLEX_I value is used
   - Under Voltage Lock Out (setUnderVoltageLimit) 5-20%
   - Over Voltage Lock Out (setUpperVoltageLimit) 5-20%
  */
  usb.setVoltage(3, 20.0);
  usb.setCurrent(3, 5.0);
  usb.setLowerVoltageLimit(3, 20);
  usb.setUpperVoltageLimit(3, 20);

  /* Flexible current value common to all PDOs */
  usb.setFlexCurrent(3.0);

  /* Unconstrained Power bit setting in capabilities message sent by the sink */
[... 100 lines cut ...]
  Serial.print("Request Source Current: ");
  Serial.println(usb.getReqSrcCurrent());
}

void loop()
{
  pdo_number = usb.getPdoNumber();
  Serial.print("PDO Number: ");
  Serial.println(pdo_number);
  
  voltage = usb.getVoltage(pdo_number);
  Serial.print("Voltage3 (V): ");
  Serial.println(voltage);
    
  delay(1000);
}`

After running this code on the RedBoard, and then pressing the reset button on the PowerDeliveryBoard, the orange LED for PDO3 on the PowerDeliverBoard lights up and on the Serial Monitor I get PDO3 with 20V/5A. So far so good.

When I detach the RedBoard from the PowerDeliveryBoard, the LED is still orange, but with less luster. Then when I detach the USB-C connector coming from the Anker PowerBank 737 from the PowerDeliveryBoard, the PDO3 LED goes dark, as it should (no power).
When I reconnect the USB-C coming from the Anker PowerBank 737, I would expect the PowerDeliveryBoard to cold start and then go into default mode, which I thought would now be the programmed PDo3. However the LED fro PDO3 stays dark and the PowerDeliveryBoard only delivers 5V. When I reattach the RedBoard to the PowerDeliveryBoard, and then press the reset button on the PowerDeliveryBoard, the PDO3 LED on the PowerDeliveryBoard lights up orange again.

I thought that the code would program the NVM with the new PDO values, and then when the PowerDeliveryBoard boots, it would read those values into the I2C and automatically switch to PDO3, with the LED going orange.

What am I doing wrong here?