sparkfun/SparkFun_STUSB4500_Arduino_Library #9: PDO values are not being stored in NVM

faircm2, 2023-11-28:
I have the following problem. I set the PDO values for PDO2 and PDO3 to 15V/3A and 20V/5A resp., using a modified version of Example2-SetParameters.ino:

`#include <Wire.h>
#include <SparkFun_STUSB4500.h>

STUSB4500 usb;
double voltage;
int pdo_number;

void setup() 
{
  Serial.begin(115200);
  Wire.begin(); //Join I2C bus
  
  delay(500);
  
  /* The Power Delivery board uses the default settings with address 0x28 using Wire.
  
     Opionally, if the address jumpers are modified, or using a different I2C bus,
     these parameters can be changed here. E.g. usb.begin(0x29,Wire1)
  
     It will return true on success or false on failure to communicate. */
  if(!usb.begin())
  {
    Serial.println("Cannot connect to STUSB4500.");
    Serial.println("Is the board connected? Is the device ID correct?");
    while(1);
  }
  
  Serial.println("Connected to STUSB4500!");
  delay(100);

  /* Set Number of Power Data Objects (PDO) 1-3 */
  usb.setPdoNumber(3);

  /* PDO1
   - Voltage fixed at 5V
   - Current value for PDO1 0-5A, if 0 used, FLEX_I value is used
   - Under Voltage Lock Out (setUnderVoltageLimit) fixed at 3.3V
   - Over Voltage Lock Out (setUpperVoltageLimit) 5-20%
  */
  usb.setCurrent(1, 1.5);
  usb.setUpperVoltageLimit(1, 5);

  /* PDO2
   - Voltage 5-20V
   - Current value for PDO2 0-5A, if 0 used, FLEX_I value is used
   - Under Voltage Lock Out (setUnderVoltageLimit) 5-20%
   - Over Voltage Lock Out (setUpperVoltageLimit) 5-20%
  */
  usb.setVoltage(2, 15.0); //10
  usb.setCurrent(2, 3.0); //5.0
  usb.setLowerVoltageLimit(2, 20);
  usb.setUpperVoltageLimit(2, 20);

  /* PDO3
   - Voltage 5-20V
   - Current value for PDO3 0-5A, if 0 used, FLEX_I value is used
   - Under Voltage Lock Out (setUnderVoltageLimit) 5-20%
   - Over Voltage Lock Out (setUpperVoltageLimit) 5-20%
  */
  usb.setVoltage(3, 20.0);
  usb.setCurrent(3, 5.0);
  usb.setLowerVoltageLimit(3, 20);
  usb.setUpperVoltageLimit(3, 20);

  /* Flexible current value common to all PDOs */
  usb.setFlexCurrent(3.0);

  /* Unconstrained Power bit setting in capabilities message sent by the sink */
  usb.setExternalPower(false);

  /* USB 2.0 or 3.x data communication capability by sink system */
  usb.setUsbCommCapable(false);

  /* Selects POWER_OK pins configuration
     0 - Configuration 1
     1 - No applicable
     2 - Configuration 2 (default)
     3 - Configuration 3
  */
  usb.setConfigOkGpio(2);

  /* Selects GPIO pin configuration
     0 - SW_CTRL_GPIO
     1 - ERROR_RECOVERY
     2 - DEBUG
     3 - SINK_POWER
  */
  usb.setGpioCtrl(0);

  /* Selects VBUS_EN_SNK pin configuration */
  usb.setPowerAbove5vOnly(true);

  /* In case of match, selects which operating current from the sink or the
     source is to be requested in the RDO message */
  usb.setReqSrcCurrent(true);

  /* Write and save settings to STUSB4500 */
  usb.write();
  
  /* Read the NVM settings to verify the new settings are correct */
  usb.read();

  Serial.println("New Parameters:\n");

  /* Read the Power Data Objects (PDO) highest priority */
  Serial.print("PDO Number: ");
  Serial.println(usb.getPdoNumber());

  /* Read settings for PDO1 */
  Serial.println();
  Serial.print("Voltage1 (V): ");
  Serial.println(usb.getVoltage(1));
  Serial.print("Current1 (A): ");
  Serial.println(usb.getCurrent(1));
  Serial.print("Lower Voltage Tolerance1 (%): ");
  Serial.println(usb.getLowerVoltageLimit(1));
  Serial.print("Upper Voltage Tolerance1 (%): ");
  Serial.println(usb.getUpperVoltageLimit(1));
  Serial.println();

  /* Read settings for PDO2 */
  Serial.print("Voltage2 (V): ");
  Serial.println(usb.getVoltage(2));
  Serial.print("Current2 (A): ");
  Serial.println(usb.getCurrent(2));
  Serial.print("Lower Voltage Tolerance2 (%): ");
  Serial.println(usb.getLowerVoltageLimit(2));
  Serial.print("Upper Voltage Tolerance2 (%): ");
  Serial.println(usb.getUpperVoltageLimit(2));
  Serial.println();

  /* Read settings for PDO3 */
  Serial.print("Voltage3 (V): ");
  Serial.println(usb.getVoltage(3));
  Serial.print("Current3 (A): ");
  Serial.println(usb.getCurrent(3));
  Serial.print("Lower Voltage Tolerance3 (%): ");
  Serial.println(usb.getLowerVoltageLimit(3));
  Serial.print("Upper Voltage Tolerance3 (%): ");
  Serial.println(usb.getUpperVoltageLimit(3));
  Serial.println();

  /* Read the flex current value */
  Serial.print("Flex Current: ");
  Serial.println(usb.getFlexCurrent());

  /* Read the External Power capable bit */
  Serial.print("External Power: ");
  Serial.println(usb.getExternalPower());

  /* Read the USB Communication capable bit */
  Serial.print("USB Communication Capable: ");
  Serial.println(usb.getUsbCommCapable());

  /* Read the POWER_OK pins configuration */
  Serial.print("Configuration OK GPIO: ");
  Serial.println(usb.getConfigOkGpio());

  /* Read the GPIO pin configuration */
  Serial.print("GPIO Control: ");
  Serial.println(usb.getGpioCtrl());

  /* Read the bit that enables VBUS_EN_SNK pin only when power is greater than 5V */
  Serial.print("Enable Power Only Above 5V: ");
  Serial.println(usb.getPowerAbove5vOnly());
  
  /* Read bit that controls if the Source or Sink device's 
     operating current is used in the RDO message */
  Serial.print("Request Source Current: ");
  Serial.println(usb.getReqSrcCurrent());
}

void loop()
{
  pdo_number = usb.getPdoNumber();
  Serial.print("PDO Number: ");
  Serial.println(pdo_number);
  
  voltage = usb.getVoltage(pdo_number);
  Serial.print("Voltage3 (V): ");
  Serial.println(voltage);
    
  delay(1000);
}`

After running this code on the RedBoard, and then pressing the reset button on the PowerDeliveryBoard, the orange LED for PDO3 on the PowerDeliverBoard lights up and on the Serial Monitor I get PDO3 with 20V/5A. So far so good.

When I detach the RedBoard from the PowerDeliveryBoard, the LED is still orange, but with less luster. Then when I detach the USB-C connector coming from the Anker PowerBank 737 from the PowerDeliveryBoard, the PDO3 LED goes dark, as it should (no power).
When I reconnect the USB-C coming from the Anker PowerBank 737, I would expect the PowerDeliveryBoard to cold start and then go into default mode, which I thought would now be the programmed PDo3. However the LED fro PDO3 stays dark and the PowerDeliveryBoard only delivers 5V. When I reattach the RedBoard to the PowerDeliveryBoard, and then press the reset button on the PowerDeliveryBoard, the PDO3 LED on the PowerDeliveryBoard lights up orange again.

I thought that the code would program the NVM with the new PDO values, and then when the PowerDeliveryBoard boots, it would read those values into the I2C and automatically switch to PDO3, with the LED going orange.

What am I doing wrong here?

ardnew, 2023-11-28:
You should try flashing `Example1-ReadParameters.ino` to your RedBoard to inspect what was actually written to NVM. The contents of NVM will not be different from whatever was written using your modified example sketch. And please post the output you get on the serial monitor from the `Example1` sketch.

This way you can verify the PDO values you set are being saved correctly, and that the problem is something else.

faircm2, 2023-11-28:
Thanks for your quick answer. I was reading some of the other posts, and found this entry by awende:
https://github.com/sparkfun/SparkFun_STUSB4500_Arduino_Library/issues/3#issuecomment-784356538

> I ran into an issue where one or more of the NVM bytes would have the wrong value written to them. As a result, you could change the PDO settings, but the board would stay at 5V. 

Lower down in the thread, there is a sample program by infamy called [USBC_PPv1.txt].
I ran that first with usb.write(DEFAULT), and then uncommenting the new PDO-entries, which I modified for myself, I ran this code with usb.write().
Now the orange LED turns on when I plug the Power Delivery Device into the PowerBank. So it seems that the modifications were not actually written into the NVM, even though the values which I had tried writing in were read out correctly using the sample program under Example1-ReadParameters.

closed 2023-11-28 by faircm2